rocketproplab/Base11-FC #10: Valve Commands

Pumuckl007, 2019-05-05:
Create commands that toggle valves.
**Engine On**
Sets the valves to such a state the O2 and RP1 are flowing.
**Engine Off**
Sets the valves in such a state that no O2 or RP1 are flowing.
**Flush Tubes**
Sets the valves such that the tubes to the engine are flushed with Helium.
**Toggle Valve**
Toggles a single valve

Pumuckl007, 2019-10-22:
This issue is available. Feel free to work on it. If you have any questions don't hesitate to ask.

scifi4life, 2019-10-22:
When constructing commands for the valves, should I leave the body/arguments of the commands empty/commented, reference a to be created subsystem for valves or directly reference the solenoids for the valves (assuming you plan to use the same subsystem/solenoid setup used for the parachute subsystem)?

Pumuckl007, 2019-10-25:
Sorry for the late reply. You can create stubs for any methods you need that are not directly related to the issue. Then you can extend the subsystem for unit testing purposes. For instance if you needed a valve subsystem you could create one with empty methods. Then you can extend it with a test version of the subsystem such as the private inner class TestSolenoid in TestParachuteSubsystem.java .

closed 2020-05-02 by Pumuckl007